platformio/platformio-atom-ide #2404: InstallationManager

tigertim08, 2020-06-16:
Error: Could not find a version that satisfies the requirement platformio (from versions: ) No matching distribution found for platformio Error: Could not install PlatformIO Core: Command '['/Users/Tim/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1 -> Installer version: 0.3.2 Platform: Darwin-19.4.0 Python version: 3.5.2 (v3.5.2:4def2a2901a5, Jun 26 2016, 10:47:25) [GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] Python path: /usr/local/bin/python3 Creating a virtual environment at /Users/Tim/.platformio/penv Updating Python package manager (PIP) in a virtual environment Virtual environment has been successfully created! Installing PlatformIO Core Collecting platformio Could not fetch URL https://pypi.python.org/simple/platformio/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:645) - skipping

ivankravets, 2020-06-16:
> Python version: 3.5.2

Please reinstall your Python following this guide for macOS: https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

Restart Atom. Does it work now?

tigertim08, 2020-06-16:
Thanks, that worked :)

closed 2020-06-16 by tigertim08

ivankravets, 2020-06-16:
Happy coding with PlatformIO!