platformio/platformio-atom-ide #2411: PlatformIO IDE installation has been suspended

Majd949, 2020-06-20:
Immediately after installing PlatformIO-ide Atom freezes and asks to force close.
After restarting it "twice" it says "Seems like PlatformIO IDE installer is already started in another window.
And I can't open anything from the PlatformIO menu. Not even the Documentation
tried restarting and reinstalling.
Running it on Windows 10

AquaGrow, 2020-06-21:
I can confim this issue!
I uninstalled Atom becasue of some issues and deleted folders .atom, .platformio, everything atom related under AppData local/roaming. So It can be called fresh installation. After installing atom, everything looks fine. Restart PC, open Atom, Installing Platformio package forces error "Editor not responding. Wait or force close". Tried 3 times with same result.
Deleting .atom/packages/platformio-ide I can restat and use atom as normal.

gnu-emu, 2020-06-22:
Hi,
I had same issues on Ubuntu and macos. I managed to get rid of the freeze by removing ~/.atom/packages/platformio-ide folder and did a:

apm install platformio-ide@2.6.0

after that atom starts and installs the updates to version 2.7.1 (if autoupdate is active) :+1:

gnu-emu, 2020-06-22:
Seems to be the same issue: https://github.com/platformio/platformio-atom-ide/issues/2412

expert3411, 2020-06-22:
that actually did work i just had to wait on the force quit screen for a bit. unfortunately it still dint fix my issue of loading my skr1.4 with marlin

Majd949, 2020-06-22:
> Hi,
> I had same issues on Ubuntu and macos. I managed to get rid of the freeze by removing ~/.atom/packages/platformio-ide folder and did a:
> 
> apm install platformio-ide@2.6.0
> 
> after that atom starts and installs the updates to version 2.7.1 (if autoupdate is active) 👍

That worked for me as well on windows 10.
Thanks!

condac, 2020-06-22:
I have this issue also on fully upgraded Arch linux system. 
`apm install platformio-ide@2.6.0` worked

mtongnz, 2020-07-25:
Confirm this works on Win10.  I had to force close Atom first.
I then uninstalled Platformio via cmd prompt:
`apm deinstall platformio-ide`

idumzaes, 2020-08-02:
Running macOS 10.14.6 and confirmed fix.
`apm deinstall platformio-ide
`apm install platformio-ide@2.6.0

arwooldridge, 2020-08-19:
tried apm deinstall platformio-ide, and apm install platformio-ide@2.6.0,  but get apm not  recognised as internal or external command. windows 10. what do i need to install?

closed 2020-09-14 by ivankravets