platformio/platformio-atom-ide #2412: platformio-ide crashes atom

expert3411, 2020-06-22:
[image]
im very new to all things coding and im trying to get marlin on my skr1.4 for my ender 3. Im running on the newst version of windows and every time platformio is installed it says "editor is not responding" and then asks if i want to "force close". I've done the usual thing and uninstalled it and atom then re downloading but nothing happens. I've even tried it on my laptop but the same issue occurred

gnu-emu, 2020-06-22:
Hi,
I had same issues on Ubuntu and macos. I managed to get rid of the freeze by removing ~/.atom/packages/platformio-ide folder and did a: 

apm install platformio-ide@2.6.0

after that atom starts and installs the updates to version 2.7.1 (if autoupdate is active) :+1:

gnu-emu, 2020-06-22:
Seems to be the same issue: https://github.com/platformio/platformio-atom-ide/issues/2411

expert3411, 2020-06-22:
like i said im very new at this what exactly Nisan apm install?

expert3411, 2020-06-22:
spoke to soon still freezing on me on version 2.6.0

TR7Mike, 2020-08-04:
I looked to move over from Visual Code to Atom for Arduino dev. So a completely fresh install on Win10 that's never seen Atom before. As soon as I install PlatformIO it hangs the main UI, but the menus work. Every now and again the Wait/Close error appears as in the OP.

I've tried on both my main desktop and laptop with exactly the same issue.

No way past other than to remove the Atom user folders (AppData Local and Roaming), but then can't access PlatformIO.

SarenT, 2020-08-28:
Same here on Ubuntu 18.04.

jmpzero, 2020-09-13:
I am having the same problem on a fresh Windows 10 Pro and a fresh installation of Atom (1.51.0). Whenever I install only the platformio-ide, the Atom GUI freezes after some time, and I need to restart it. I can only access the menu bar and that's it.

*My solution:* I found the config.js file, there are several packages in the ATOM_DEPENDENCIES, which I installed by hand in Atom. After the manual installation I installed PlatformIO from within Atom and all of a sudden I could see a) the update window, where PlatformIO tells me what it is currently installing and b) I could use Atom after a restart.

This solution is more a workaround or diry hack, but it works so far (at least for me).

ivankravets, 2020-09-14:
Please update to 2.7.2. Sorry for the issue.

closed 2020-09-14 by ivankravets

ptsonkov, 2020-11-05:
@ivankravets version 2.7.2 crash as well, most of the time ask to force close Atom, but sometimes (rare) there is different message - Tell that Platformio is build on the top of Python 3.+ but can't find Python (3 options available: Install python, show where is python.exe and ignore message)
Last stable version for me is 2.6.0

ChrisLomont, 2021-01-27:
Facing same issue, Win 10, VS Code 1.52.1. After spending far too long trying to get this to work, rebooting, fiddling with all sorts of ad-hoc solutions from the web, and seeing how long this issue has been affecting users, I'm moving on to other solutions. PlatformIO is too unusable to get any work done in. I'd be afraid at this point if it did get working a VS code or platform IO update would simply make work impossible again.