platformio/platformio-atom-ide #2414: Can't get PlatformIO to work

kaffeeringe, 2020-06-22:
### Configuration

**Operating system**: Windows 10 / Ubuntu 20.04

**PlatformIO Version** (`platformio --version`): 2.7.1

### Description of problem
I tried PlatformIO IDE with Visual Studio, CLion and Atom on Windows 10. And only with Atom on Ubuntu 20.04. But when I install it, CPU load goes way up and the editor stops working. 

#### Steps to Reproduce

1. Fresh Atom install on Ubuntu 20.02
2. Installed PlatformIO with apm install platformio-ide
3. Restart Atom

### Actual Results
First restart of Atom: CPU load goes to 106% - I get asked if I want to force close Atom, because it's not responding

Second restart: "PlatformIO IDE installation has been suspended. Seems like PlatformIO IDE Installer is already started in another window."

### Expected Results
Atom should keep on working

gnu-emu, 2020-06-23:
Hi,
I had same issues on Ubuntu and macos. I managed to get rid of the freeze by removing ~/.atom/packages/platformio-ide folder and did a:

apm install platformio-ide@2.6.0

after that atom starts and installs the updates to version 2.7.1 (if autoupdate is active) :+1: 

Seems to be the sam issue: https://github.com/platformio/platformio-atom-ide/issues/2411

kaffeeringe, 2020-06-25:
@p-vernaeckt: It's not just Atom. It's not just Ubuntu. I installed PlatformIO under Ubuntu/Atom as well as under Windows 10 with Atom / VSCode / CLion. Always the same result.

closed 2020-09-14 by ivankravets