platformio/platformio-atom-ide #2439: Platformio-ide crashs the Editor

ITCrowdIOB, 2020-08-12:
Platformio-ide crashs the Editor. Deinstall Atom and install don't solve the problem.

ErrOzz, 2020-08-12:
Confirm. Win10x64. Atom1.50.0x64. Platformio-IDE2.7.1

ITCrowdIOB, 2020-08-12:
Win10x64  pro  18363.1016

Atom 1.50.0 x64

Platformio-IDE2.7.1

clemans, 2020-08-12:
Same. 
`~$: cat /etc/os-release | grep -i "PRETTY NAME"` Pop!_OS 20.04 LTS
`~$: atom --version` Atom    : 1.49.0 | Electron: 5.0.13 | Chrome  : 73.0.3683.121 | Node    : 12.0.0
Platformio-ide  : 2.7.1

ErrOzz, 2020-08-13:
At the moment I have temporary fixed this by installing the previous platformio-ide version 2.6.0: apm install platformio-ide@2.6.0

Kubis12, 2020-08-15:
Could you tell me please, how can I instal platformio-ide version 2.6.0?

clemans, 2020-08-15:
> apm install platformio-ide@2.6.0

This worked for me as well. Thank you @ErrOzz . Looking forward for the @latest to getting fixed.

ErrOzz, 2020-08-16:
> Could you tell me please, how can I instal platformio-ide version 2.6.0?
You need to uninstall platformio-ide 2.7.1. Just type in the command line:
apm uninstall platformio-ide@2.7.1
You can clarify the names of the installed packages with folowing command:
apm list
When the uninstall is complete, just install previous platformio package using the command:
apm install platformio-ide@2.6.0
Launch Atom for the platformio to finish installing the required components.

closed 2020-09-14 by ivankravets