platformio/platformio-atom-ide #2442: InstallationManager

trevor-sonic, 2020-08-27:
I have impression that I'll never be able to install PlatformIO on Atom (or Visual Code the same).
Every time lunch Atom, it tries to update and gives this error.
One time I saw a green successful message, just after re-lunch the app, same miserable scenario, error.

```Error: ENOENT: no such file or directory, mkdir '~/.platformio/.cache' at Object.mkdirSync (fs.js:778:3) at Object.e.getCacheDir (/Users/theuser/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:5337) at l (/Users/theuser/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:877949) at e.callInstallerScript (/Users/theuser/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:32688) at t.value (/Users/theuser/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:27194) at async Promise.all (index 3)```

ivankravets, 2020-09-14:
Please update to 2.7.2. Sorry for the issue.

closed 2020-09-14 by ivankravets

trevor-sonic, 2020-09-18:
OMG 2.7.2 could install and is working!
I was about losing my hope, finally it works. 
Thanks.

ivankravets, 2020-09-18:
Happy coding with PlatformIO! ☺️