platformio/platformio-atom-ide #2444: Editor Window Has Become Unresponsive macos

utdrmac, 2020-09-06:
Just downloaded the latest Atom, and tried to install the latest platformIO. Atom keeps displaying the noted error.

```
$ atom --version
Atom    : 1.50.0
Electron: 5.0.13
Chrome  : 73.0.3683.121
Node    : 12.0.0
```

If I `apm remove platformio-ide`, then Atom works again. Others posted the solution to install a specific, previous version, of platformIO, 2.6.0

`apm install platformio-ide@2.6.0`

Atom works now without issue and I'm able to edit my projects with platformIO.

ivankravets, 2020-09-14:
Please update to 2.7.2. Sorry for the issue.

closed 2020-09-14 by ivankravets

mf-1, 2020-09-14:
Issue is still happen....
[image]

https://github.com/platformio/platformio-atom-ide/issues/2445#issuecomment-692101902

commit 8da87e59 (2020-09-14): Fixed an issue when Atom has become unresponsive // Resolve #2444