platformio/platformio-atom-ide #2445: Uncaught Error: ENOENT: no such file or directory, lstat '/usr/bin/atom'

MPForsyth, 2020-09-07:
[Enter steps to reproduce:]
click on Atom tab
1. ...
2. ...

**Atom**: 1.50.0 x64
**Electron**: 5.0.13
**OS**: Ubuntu 20.04.1
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.7.1

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, lstat '/usr/bin/atom'

```
At fs.js:122

Error: ENOENT: no such file or directory, lstat '/usr/bin/atom'
    at Proxy.realpathSync (fs.js:1512:7)
    at Proxy.fs.realpathSync (electron/js2c/asar.js:300:40)
    at /packages/platformio-ide/lib/utils.js:102:25
    at Object.getActivePioProject (/packages/platformio-ide/lib/utils.js:106:6)
    at Object.highlightActiveProject (/packages/platformio-ide/lib/maintenance.js:103:19)
    at doHighlight (/packages/platformio-ide/lib/main.js:305:43)
    at Function.simpleDispatch (/snap/atom/259/usr/share/atom/resources/app/static/<embedded>:11:1190202)
    at Emitter.emit (/snap/atom/259/usr/share/atom/resources/app/static/<embedded>:11:1191643)
    at /snap/atom/259/usr/share/atom/resources/app/static/<embedded>:11:426378
```

### Commands

```
  2x -1:34 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-solarized-dark-syntax.theme-one-dark-ui)
```

### Non-Core Packages

```
autocomplete-clang 0.12.1 
build 0.70.0 
busy-signal 2.0.1 
file-icons 2.1.38 
intentions 1.1.5 
language-ini 1.23.0 
linter 2.3.1 
linter-gcc 0.9.0 
linter-gcc2 0.8.8 
linter-ui-default 1.8.1 
my-package 0.0.0 
platformio-ide 2.7.1 
platformio-ide-debugger 1.2.6 
platformio-ide-terminal 2.10.0 
tool-bar 1.2.0 
```

mf-1, 2020-09-11:
I've the same issue....

```
Error: ENOENT: no such file or directory, lstat '/usr/bin/atom'
    at Proxy.realpathSync (fs.js:1512:7)
    at Proxy.fs.realpathSync (electron/js2c/asar.js:300:40)
    at /home/markus/.atom/packages/platformio-ide/lib/utils.js:102:25
    at Object.getActivePioProject (/home/markus/.atom/packages/platformio-ide/lib/utils.js:106:6)
    at Object.highlightActiveProject (/home/markus/.atom/packages/platformio-ide/lib/maintenance.js:103:19)
    at doHighlight (/home/markus/.atom/packages/platformio-ide/lib/main.js:305:43)
    at Function.simpleDispatch (/snap/atom/260/usr/share/atom/resources/app/static/<embedded>:11:1190297)
    at Emitter.emit (/snap/atom/260/usr/share/atom/resources/app/static/<embedded>:11:1191738)
    at /snap/atom/260/usr/share/atom/resources/app/static/<embedded>:11:426473

```

I tried to start a new window / new instance of Atom including Platform IO using Ubuntu 20.20.

ivankravets, 2020-09-14:
https://github.com/platformio/platformio-atom-ide/issues/2444

closed 2020-09-14 by ivankravets

mf-1, 2020-09-14:
Issue is still happens using PIO 2.7.2 

> I tried to start a new window / new instance of Atom including Platform IO using Ubuntu 20.20.