platformio/platformio-atom-ide #2456: platformio-ide-terminal do not start

ptsonkov, 2020-11-07:
Terminal and serial console do not start due to following error:
```
<embedded>:14 Cannot find module '../build/Debug/pty.node'
Require stack:
- /home/bubba/.atom/packages/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch/lib/unixTerminal.js
- /home/bubba/.atom/packages/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch/lib/index.js
- /home/bubba/.atom/packages/platformio-ide-terminal/lib/process.coffee
- /usr/share/atom/resources/app.asar/src/task-bootstrap.js Error: Cannot find module '../build/Debug/pty.node'
Require stack:
- /home/bubba/.atom/packages/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch/lib/unixTerminal.js
- /home/bubba/.atom/packages/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch/lib/index.js
- /home/bubba/.atom/packages/platformio-ide-terminal/lib/process.coffee
- /usr/share/atom/resources/app.asar/src/task-bootstrap.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/bubba/.atom/packages/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch/lib/unixTerminal.js:29:11)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.value [as .js] (/usr/share/atom/resources/app.asar/src/compile-cache.js:255:23)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/bubba/.atom/packages/platformio-ide-terminal/node_modules/node-pty-prebuilt-multiarch/lib/index.js:13:20)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.value [as .js] (/usr/share/atom/resources/app.asar/src/compile-cache.js:255:23)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/bubba/.atom/packages/platformio-ide-terminal/lib/process.coffee:1:7)
    at Object.<anonymous> (/home/bubba/.atom/packages/platformio-ide-terminal/lib/process.coffee:1:1)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.value [as .coffee] (/usr/share/atom/resources/app.asar/src/compile-cache.js:255:23)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/usr/share/atom/resources/app.asar/src/task-bootstrap.js:90:17)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
(anonymous) @ <embedded>:14
(anonymous) @ <embedded>:14
simpleDispatch @ <embedded>:11
emit @ <embedded>:11
(anonymous) @ <embedded>:14
emit @ events.js:200
emit @ internal/child_process.js:876
processTicksAndRejections @ internal/process/task_queues.js:84
<embedded>:14 Cannot access 'handler' before initialization ReferenceError: Cannot access 'handler' before initialization
    at process.<anonymous> (/usr/share/atom/resources/app.asar/src/task-bootstrap.js:69:20)
    at process.emit (events.js:200:13)
    at emit (internal/child_process.js:876:12)
    at processTicksAndRejections (internal/process/task_queues.js:84:9)
(anonymous) @ <embedded>:14
(anonymous) @ <embedded>:14
simpleDispatch @ <embedded>:11
emit @ <embedded>:11
(anonymous) @ <embedded>:14
emit @ events.js:200
emit @ internal/child_process.js:876
processTicksAndRejections @ internal/process/task_queues.js:84
```
Workstation: Fedora33

$ apm -v
apm  2.5.2
npm  6.14.8
node 12.4.0 x64
atom 1.52.0
python 3.9.0
git 2.28.0

$ apm list | grep -i platformio
├── platformio-ide@2.7.2
├── platformio-ide-debugger@1.2.6
├── platformio-ide-terminal@2.10.0

Please let me know if you need some more info

BR,
Peter

ptsonkov, 2020-11-07:
some more info. I try to make/compile by hand, but it return following error (attached is full log)
```
make: *** [pty.target.mk:109: Release/obj.target/pty/src/unix/pty.o] Error 1
make: Leaving directory '/home/bubba/.atom/packages/platformio-ide-terminal/node_modules/term.js/node_modules/pty.js/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 5.8.17-300.fc33.x86_64
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/bubba/.atom/packages/platformio-ide-terminal/node_modules/term.js/node_modules/pty.js
gyp ERR! node -v v14.14.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! pty.js@0.3.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the pty.js@0.3.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/bubba/.npm/_logs/2020-11-07T20_46_43_953Z-debug.log
```
```
[bubba@host ~/.atom/packages/platformio-ide-terminal/node_modules/term.js  ]$ 
```
[2020-11-07T20_46_43_953Z-debug.log](https://github.com/platformio/platformio-atom-ide/files/5505410/2020-11-07T20_46_43_953Z-debug.log)

ptsonkov, 2020-11-07:
last update - downgrade to Atom 1.42.0 fix the issue, all work fine

ivankravets, 2020-12-12:
Fixed in https://github.com/platformio/platformio-atom-ide-terminal/releases/tag/v2.10.1

Please update `platformio-ide-terminal` to the latest version.

closed 2020-12-12 by ivankravets