platformio/platformio-atom-ide #2457: Platformio-ide

Blakinas, 2020-11-08:
instaliing whole week, when it will finish? Please help.
if i restart Atom, the same keeps tring after some time. but never finished.
[image]
[image]

my email is tomas.sliauteris@gmail.com

Winch-, 2020-11-22:
I have the same issue, install hangs with installing dependences platformio toolbar is there but does not work. also small flashing icon at bottom of screen, checking PIO core.
tried all sorts of suggestions from web but no good. *my coding skills are basic so some suggestions were beyond my skill set*. surely this is a flagship app and should work without bugs. I am following online coding examples where Atom is used and it seems like a great tool so why does it work for some and not others.

n3o23, 2020-11-24:
Same issue here. Been stuck installing all day.
I heard if you install the older version it will work just fine.  I am researching and will provide an update once I get it fixed.

B0bb3rl3, 2020-12-01:
I've the same problem. Tried a workaround. But it still hangs up... Hope there will be a solution for this problem soon.

maxgerhardt, 2020-12-11:
@ivankravets can you have a look at this? Also just popped up in https://community.platformio.org/t/platformio-ide-install-failure-under-atom-editior-under-windows/17703

ptmartellock, 2020-12-11:
LOL, the last one's reference only points to this thread... circular no answers.

ivankravets, 2020-12-12:
Fixed in https://github.com/platformio/platformio-atom-ide-terminal/releases/tag/v2.10.1

Please update `platformio-ide-terminal` to the latest version.

closed 2020-12-12 by ivankravets