platformio/platformio-atom-ide #2461: Atom: "Try PlatformIO IDE for VSCode" wont stop popping up

m-r-m-s, 2020-12-10:
Hi, I am using Atom with the PlatformIO IDE installed and even after clicking “Do not show this message”, or "Remind me Later", the  “Try PlatformIO IDE for VSCode” pops up every time I open Atom:

[image]

How do I disable this window and prevent it from popping up every time I open Atom?

This is the version of Atom I am using:
[image]

evlquaker, 2020-12-11:
I also have this problem. It is very annoying

c-o-m-m-a-n-d-e-r, 2021-01-04:
having this too @ osx 1.53.0

SeaGull711, 2021-01-05:
Same here!  Sooo annoying!

evlquaker, 2021-01-05:
I fixed it on my computer using a really half-assed approach.
File -> settings -> packages
find platformio-ide in the list and click settings
click View Code
on the left panel platformio-ide -> lib -> maintenance.js

find the line "export function notifyExtensionIsDeprecated() {" which should be somewhere around line #82. 
write "return;" right under this line and save the file. No more annoying popup.

The actual problem is related to the local storage variable "platformio-ide:extension-deprecated" not being set properly. Hence the line "if (localStorage.getItem('platformio-ide:extension-deprecated') === '1') {
    return;
  }" always evaluates to false. I don't know the proper way to fix it, but I fixed my problem with it.

m-r-m-s, 2021-01-06:
`return;` is already there on mine unfortunately...

`export function notifyExtensionIsDeprecated() {
  if (localStorage.getItem('platformio-ide:extension-deprecated') === '1') {
    return;
  }`

evlquaker, 2021-01-06:
change that to:

export function notifyExtensionIsDeprecated() { 
return;
if (localStorage.getItem('platformio-ide:extension-deprecated') === '1') { return; }

the idea is you are just writing return before the if statement so it skips the whole thing

m-r-m-s, 2021-01-06:
Ahhh yep - thank you! I was toggling back and forth between the browser and Atom and now I see. Thanks for the additional clarification

evlquaker, 2021-01-06:
np, glad it helped :-)

c-o-m-m-a-n-d-e-r, 2021-01-06:
Thanks Bro :-)

Fxuko, 2021-01-10:
Esa funcion no afecta al programa.  Yo la he comentado con ---- > /** y **/   .   Hay que poner antes de la variable y al final donde termina la funcion para dejarla sin uso.
238 / 5000
Resultados de traducción
This function does not affect the program. I have commented it with ----> / ** and ** /. You have to put before the variable and at the end where the function ends to leave it unused. It works perfectly.

/ **
export function ......
...
}
** /

Fxuko, 2021-01-10:
238 / 5000
Resultados de traducción
This function does not affect the program. I have commented it with ----> / ** and ** /. You have to put before the variable and at the end where the function ends to leave it unused. It works perfectly.

/ **
export function ......
...
}
** /

evlquaker, 2021-01-10:
@Fxuko 
I am not sure I understand what you are saying. Are you suggesting the whole notifyExtensionIsDeprecated function should be commented out? The function is still being called. If it is commented out without removing all function calls there will be an error. It is possible you won't even see the error, but it still exists. This is not a good thing to do.

Fxuko, 2021-01-10:
That function is called at the beginning of the program only. So far it has not given me problems.