platformio/platformio-atom-ide #2463: Error installing PIO

HugoGroen, 2021-01-07:
When installing PIO, I get the same error every time, no matter what solution I try:
`Error: Error: Compatible PlatformIO Core not found.
Reason: PlatformIO executable not found in `C:\Users\Hugo Groeneveld\.platformio\penv\Scripts`
	at c:\Users\Hugo Groeneveld\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879275	at ChildProcess.c (c:\Users\Hugo Groeneveld\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)	at ChildProcess.emit (events.js:223:5)	at ChildProcess.cp.emit (c:\Users\Hugo Groeneveld\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)	at maybeClose (internal/child_process.js:1021:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
t.log @ console.ts:112
`
And
`Error: Traceback (most recent call last):   File "C:\Users\Hugo Groeneveld\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>     main()   File "C:\Users\Hugo Groeneveld\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main     bootstrap()   File "C:\Users\Hugo Groeneveld\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap     pioinstaller.__main__.main()   File "C:\Users\Hugo Groeneveld\.platformio\.cache\tmp\.piocore-installer-w53wrrkj\tmp815vkpj9\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main   File "C:\Users\Hugo Groeneveld\.platformio\.cache\tmp\.piocore-installer-w53wrrkj\tmp815vkpj9\pioinstaller.zip\click\core.py...
onDidChangeNotification @ notificationsAlerts.ts:36
`
And the last one
`contextKeyService.ts:420 Element already has context attribute
w @ contextKeyService.ts:379
`

I have no idea what to make of these errors or how to fix them. I've already tried to 'Check available solutions' which leads me [here](https://github.com/platformio/platformio-core-installer/issues/221), but this fix doesn't work. Perhaps the problem is that I have multiple different versions of Python installed (2.7.15, 2.7.16, 3.8.2 & 3.9.1) and I haven't enabled 'Add to PATH' in the installation, as recommended by TH3D Studio (I'm trying to edit Marlin code for my 3D printer). Could anyone tell me how to fix this? Thanks!

daes10, 2021-02-09:
I have the same issue and every thing doesnt worked.

System: Windows 10
Tested both in VSCode and Atom. There ist the same issue.

`Error: Traceback (most recent call last):   File "C:\Users\David Trachsler\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>     main()   File "C:\Users\David Trachsler\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main     bootstrap()   File "C:\Users\David Trachsler\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap     pioinstaller.__main__.main()   File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main   File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\click\core.py...
onDidChangeNotification @ notificationsAlerts.ts:40`

`[Extension Host] Error: Traceback (most recent call last):
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv
  File "C:\Users\David Trachsler\.platformio\.cache\tmp\.piocore-installer-wddr8r92\tmp6cahttg5\pioinstaller.zip\pioinstaller\python.py", line 170, in find_compatible_pythons
  File "C:\Users\David Trachsler\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\David Trachsler\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\David Trachsler\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\David Trachsler\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 193] %1 ist keine zul�ssige Win32-Anwendung
 -> Installer version: 0.3.5
Platform: Windows-10
Python version: 3.9.1 (tags/v3.9.1:1e5d33e, Dec  7 2020, 17:08:21) [MSC v.1927 64 bit (AMD64)]
Python path: C:\Users\David Trachsler\AppData\Local\Programs\Python\Python39\python.exe
Creating a virtual environment at C:\Users\David Trachsler\.platformio\penv
	at c:\Users\David Trachsler\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879275	at ChildProcess.c (c:\Users\David Trachsler\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)	at ChildProcess.emit (events.js:315:20)	at ChildProcess.cp.emit (c:\Users\David Trachsler\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)	at maybeClose (internal/child_process.js:1021:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)`

HugoGroen, 2021-02-09:
The problem seems to lie with the fact that I, and you most likely as well, have a space in my Windows username (eg: John Doe, note the space). This seems to make either VisualStudio or POI mess up. I have found a temporary solution by using my account without a space in the name (eg: JohnDoe) with seems to make everything work. It still is a really annoying and stupid bug but I guess we have to deal with it for now.